haplokuon/netDxf #2: VertexFlags information gets lost

GustavAT, 2017-10-10:
Please refer to the following sample-file: [mp1.txt](https://github.com/haplokuon/netDxf/files/1371560/mp1.txt)

The file contains a single Polyline. The line's vertices have either the _SplineVertexFromSplineFitting_ or the _SplineFrameControlPoint_ flag. When opening my sample file, the DxfReader will eventually convert the Polyline to an LwPolyline object. The problem here is that the flags for each vertex get lost. In my application, I cannot differ between vertex types anymore.

_Tested with 2.0.3_

haplokuon, 2017-10-11:
The available options in AutoCad to fit a polyline to a curve are not implemented in netDxf, only the standard polylines are supported. Therefore the information related to this might be lost during the process.

GustavAT, 2017-10-12:
Thanks for your quick response!

I don't need to calculate the vertices of the polyline based on the control points as they are already present in the file. But when querying the LwPolyline's vertices, I get the control points and the actual polyline points mixed up.

I think that adding a flag to the LwPolylineVertex class similar to the Vertex class would solve my problem. Otherwise, would it be possible to filter out control points during the reading process if the respective flags are set?

haplokuon, 2017-11-03:
Yes, I guess there is no problem of keeping that value, whatever it is, so it is maintained when the file is saved again.

Daniel

haplokuon, 2017-11-28:
Should work with the new changes in [2017/11/28]

closed 2017-11-28 by haplokuon