haplokuon/netDxf #4: The outputstream.Position does not exist

Yahasana, 2017-10-13:
https://github.com/haplokuon/netDxf/blob/40601825513b5096604122c23edb12ea7a4fcd93/netDxf/IO/DxfWriter.cs#L392

e.g. `dxfDoc.write(context.Response.OutputStream)` will raise this error

haplokuon, 2017-11-03:
You should be able to delete that line safely, it is not really needed. Some leftover, I guess, I didn't have any issues with it, and I forgot to delete it.

In the DxfReader class there are a couple of lines like that "stream.Position = 0;", they are there for convenience. You can check a few things of the dxf without having to read the whole thing. Like checking is it is binary or a header variable, and rewinding the stream is needed in those cases.

I should let the caller of the methods that use it decide if it is needed. Since they are the responsibility of properly handling the stream is the caller not the method; just like the caller is the responsible of disposing it.

Daniel

haplokuon, 2017-11-28:
Fixed in [2017/11/28]. Even thought I have not been able to reproduce the problem, but being able to control the stream position will give you some more flexibility.

closed 2017-11-28 by haplokuon