haplokuon/netDxf #8: Interest in supporting older DXF versions

doug8679, 2017-11-13:
I have project which needs to work with DXF files, however, we cannot always ensure that the DXF is in 2000 or newer format.  Is it possible to extend the library to handle older versions of DXF in ASCII format?Is that a long-range plan for the library at all?

haplokuon, 2017-11-15:
Possible, yes it is possible, I have thought about adding the option of loading (read only) old v12 dxf files. But, sorry, I cannot tell you when it will happen.

Daniel

doug8679, 2017-11-15:
If it were a funded effort, do you think it is something you would be interested in pursuing?  We have several projects that could benefit from this library if it could read older versions of DXF as well as the ones it currently reads.

haplokuon, 2017-11-20:
A funded effort will not solve anything. if what you need is professional help, I will recommend you, it will be cheaper, to try one of the commercial options, and I am not talking about the way overpriced solution Autodesk offers.

Daniel

ZmorzynskiK, 2018-06-21:
Hi,

I have the same problem: need to load DXF R12 files, even in the most basic form (just the entities).
So I forked your repo @haplokuon and did some... hacks:
https://github.com/ZmorzynskiK/netDxf

I added `GlobalOptions.IsBackwardCompatibilityEnabled` (default false) that when set to true, then certain file processing is a little different from current. When false, then everything should be as in normal version.
Not the most elegant solution but as a quick hack it sort of works for some R12 files I've tested.

Also added some optional `ReadHex` values reading because I saw that LibreCad saves DXFs without some info (the code is there but the value is empty).