haplokuon/netDxf #9: Point style

baucez, 2017-11-14:
Is it possible to apply a point style (let say a cross) when writing a point in a document?

sorifiend, 2017-11-14:
Point styles are generally just a block of lines? You could make a block yourself out of two intersecting lines and insert it at each of your points.

haplokuon, 2017-11-15:
@baucez 
Yes, you can. It is controlled by the PdMode and PdSize properties that you can find in the DrawingVariables of the DxfDocument, those properties control the shape and size of all Point entities. Keep in mind that their values are global and will affect all points in the drawing.

@sorifiend 
Point entities are not blocks, they are different entities; but if AutoCad internally handles them like that I don't know. You could create a block to use them as a point if you like, nothing prevents you from doing that.

Daniel

baucez, 2017-11-15:
@haplokuon That's exactly what I was looking for! Thanks

closed 2017-11-15 by baucez