haplokuon/netDxf #13: Minimum vertex spacing?

jaret81, 2017-11-19:
I have an array of XY coordinates that I'm attempting to export to dxf.  The resulting shape is that of a rough circle with a circumferential resolution of about 0.003" and 4800 points.

If I reduce the resolution by about 20x I am able to create a nice dxf image (image on left).  However, any higher resolution results in garbage (image on right).  I have tried using lines, polylines, lwpolylines, and a spline with similar results.

[image]

What can I do to allow the full resolution?  Any advice you can provide would be greatly appreciated!

Thank you,
Jaret

haplokuon, 2017-11-20:
Have you tried saving your file in binary format? Keep in mind that the number of digits written, for a double, in a text based dxf is limited to 15 digits. This is a limitation of the net ToString method, you could squeeze one more using the R option but it will not change all that much. The binary format should preserve the full range of a double variable.

I am assuming you are exporting this with netDxf, other programs like AutoCad allows you to choose the number of digits written in a text dxf, if I recall correctly the maximum was 16. Now that disk space is not a problem using less that the maximum allowed is pointless. 

Daniel

jaret81, 2017-11-20:
Thank you for your prompt reply!

Yes, I am saving in binary format with netDXF.  I don't think it is a precision issue, as I only need accuracy to 3 decimals.  There seems to be some limitation with vertex proximity.

Here is the code and the array of coordinates.  You will find that unless you increment nP by 20+ the output will look like the above image on the right.

            DxfDocument dxf = new DxfDocument();
            Polyline poly = new Polyline();
            for (int nP = 0; nP < _arrScanLine.Length - 2; nP += 2)
            {
                double fX = (double)_arrScanLine[nP];
                double fY = (double)_arrScanLine[nP + 1];
                poly.Vertexes.Add(new PolylineVertex(fX, fY, 0));
            }
            poly.IsClosed = true;
            dxf.AddEntity(poly);

[XY Coordinates.txt](https://github.com/haplokuon/netDxf/files/1488638/XY.Coordinates.txt)

haplokuon, 2017-11-21:
There is nothing in the netDxf code that will limit the proximity of two vertexes, besides the obvious limitations of the double data type.

Check your XY Coordinates list there are a few vertices that are way out of the shape of the circle, if you zoom extents the whole drawing it will look like only a line has been drawn, but if you zoom close enough you will see the shape of the circle with a few vertices far to the right. For example this is the values you have on line 128 are "987986,0". I guess when you reduced the number of points you were lucky and all those points, way out of the circle shape, were left behind.

Daniel

jaret81, 2017-11-21:
That's bizarre, but you're absolutely correct.  The program that collects the data sets undefined data points to a constant int of 987987, but it has clearly been changed to 987986 somewhere along the line.  I changed my undefined catch from ==987987 to >900000, and the dxf looks great!

Thank you for your help on this.  It should have been more obvious, but I don't have Solidworks, and eDrawings wasn't allowing me to zoom in on the drawing.

Also, thank you for the excellent dll! 

Jaret

closed 2017-11-21 by jaret81