haplokuon/netDxf #14: Feature request : Support for Rounded polygons

Sumeet2490, 2017-11-21:
One of the features I'm developing for my CAD generator needs to round the corners of a polygon.
Could you please suggest how can I do this?

Thanks!

haplokuon, 2017-11-23:
You can easily build rounded polygons using the LwPolyline entity and the Bulge property of its vertices. First you will need to chamfer its corners and then set the bulge value accordingly for every other vertex.

The only entities you will find in the library are the ones present in the dxf database. This library does not pretend to solve geometric problems but to help reading and saving dxf files in a more meaningful way.

Daniel

Sumeet2490, 2017-11-23:
Thanks for your response. Could you please give a code snippet? (like a rounded square with corner radius 1)

haplokuon, 2017-11-24:
Please, try to work it out yourself first. Later, you can come back with your problems with something to show.

Daniel

closed 2018-05-04 by haplokuon