haplokuon/netDxf #17: DXF-Reader

promikeda, 2018-01-03:
Hello,

I would like to process DXF-Drawings created in SolidWorks. 

I found out that a drawing apart from the header potentially consists of inserts, blocks and geometries (circles,...). 
I parsed geometries in my drawing -> I found circles with centers that seem to be in paper coordinates since the numbers change when I move the view before exporting to dxf : 
Console.WriteLine(circ.ToString() + "(.Owner= " + circ.Owner + ") CX=" + circ.Center.X + " CY=" + 
circ.Center.Y);
Circle(.Owner= *Model_Space) CX=196,0103171922 CY=84,4324113898
Circle(.Owner= *Model_Space) CX=198,7291570945 CY=92,8001401971
Circle(.Owner= *Model_Space) CX=200,377803623 CY=101,44264796

I wanted to see the coordinates not in paper coordinates but in "view" or "part" coordinates. 
The circles are owned by a block called *Model_Space which seems to have its origin at (0,0).
 Console.WriteLine("*************************************");
            Console.WriteLine("blocks");
            foreach (netDxf.Blocks.Block block in dxfLoad.Blocks)
            {
                Console.WriteLine("Name=" + block.ToString());
                {
                    Console.WriteLine(block.ToString() + ".Origin.X=" + block.Origin.X);
                    Console.WriteLine(block.ToString() + ".Origin.Y=" + block.Origin.Y);
                    Console.WriteLine(block.ToString() + ".Origin.Z=" + block.Origin.Z);
                }
            }
*************************************
blocks
Name=*Model_Space
*Model_Space.Origin.X=0
*Model_Space.Origin.Y=0
*Model_Space.Origin.Z=0
Name=*Paper_Space
*Paper_Space.Origin.X=0
*Paper_Space.Origin.Y=0
*Paper_Space.Origin.Z=0
Name=*Paper_Space0
*Paper_Space0.Origin.X=0
*Paper_Space0.Origin.Y=0
*Paper_Space0.Origin.Z=0
Name=SW_CENTERMARKSYMBOL_0
SW_CENTERMARKSYMBOL_0.Origin.X=0
SW_CENTERMARKSYMBOL_0.Origin.Y=0
SW_CENTERMARKSYMBOL_0.Origin.Z=0
*************************************

If I subtract the Origin of the one and only Insert in the drawing I don't get correct values either.

A workaround could be a cirlcle with another linetype denoting the origin of the "view" or the part. 

Is there a better alternative ? 

best regards

M

haplokuon, 2018-01-04:
The center of the circle is defined in world coordinates, and in your case all those circles belongs to the "Model" layout and not to any Paper Space layout. The block "\*Model_Space" you see as owner of the circle is the way AutoCad internally handles the layouts, any block which name starts with '\*' is for internal use, you cannot manually create one of those.

Something else must be happening here. The position of an object is independent on the way you see it, therefore moving the view (as you say it) should not affect, in any way, the position of your circles. Are you sure you are moving the view and not the object itself? From the viewer point of view moving the world and moving the view is in practice the same thing.

What do you mean by "I wanted to see the coordinates not in paper coordinates but in "view" or "part" coordinates." All entities' coordinates are expressed either in world coordinates, global XY plane with normal (0,0,1); or object coordinates, which construction plane is defined by its normal. You can find a method to convert between world and local coordinates in the MathHelper class.

Daniel

promikeda, 2018-01-05:
Hello,

indeed I moved the position of the projection of a part (that I denoted as "view") on the paper. That changes the values X&Y of the circle.Center which are assigned to *Model_Space.
With "I want to see the coordinates in part coordinates" I meant that I wanted to see measurements related to a origin located on the part instead of coordinates related to  *Model_Space which has its origin at the paper corner. Therefore I decided to require the user to draw a centermark (that I can find with netdxf) to the projection and I calculate my coordinates by subtraction of that origin.

closed 2018-01-05 by promikeda