haplokuon/netDxf #18: [Question] Is it possible to link netDxf COM dll to Excel/VBA?

jocdiazmuic, 2018-01-05:
Hello,

For a personal project I'm working on a module that heavily relies on I/O operations with dxf files in a rather simple Excel VBA module. I tried to link a compiled netDXF DLL with COM visibility support and registering in my personal machine following [this procedure](http://www.geeksengine.com/article/create-dll.html) using Visual Studio Community 2017.

EXCEL is able to identify and reference the DLL and TLB appropriately but I can't manage to call a single function nor parameter from the netDXF , and I somehow feel uneasy that VBA Tooltips (a.k.a intellisense) are not displaying anything for netDxf's functions.

Is it possible to load and work with netDXF as a COM dll in Excel/VBA at all? I'm sure I'm missing some steps along the way and maybe this is not the intended used of the library...

P.S.
 I'm not related to the .NET programming env. but I do feel confortable with EXCEL/VBA so I decided to give this a shot, any help in this matter is greatly appreciated.

Cheers,

José D.

haplokuon, 2018-01-08:
I have not tried, but the code will not work as it is. COM has its limitations, and while for simple examples the COM interface has mechanisms to automatically generate interfaces from the existing classes, more complex libraries will not work. it will be necessary to define explicit interfaces. [Here](https://docs.microsoft.com/en-us/dotnet/framework/interop/exposing-dotnet-components-to-com) you can find more detailed information about how to expose net components to COM.

Daniel

jocdiazmuic, 2018-01-09:
Hello Daniel, thank you for your prompt response.

I was able to expose quite a bunch of the functions and classes from netDXF within the COM interface I was quite happy until I came across one of the COM limitations that you mentioned before: Static members (functions, properties and such) which I believe are rather common within .NET aren't supported within VBA at all, so functions like 'Save' seems to work ok but that leaves things like 'dxfdocument.Load' out of the reach.

I'll keep you updated if something happens. Thank you for your time and efforts!

closed 2018-05-04 by haplokuon