haplokuon/netDxf #19: UCS applied incorrectly?

Troubadix75, 2018-01-10:
Hello,

it looks like the new feature "Added the header DrawingVariables UcsOrg, UcsXDir, and UcsYDir." is applied incorrectly.

I have a DXF File here that shows all elements in the XY Plane when I open it in AutoDesk DWG TrueView. When I open it in my Program (using a copy of todays netDxf) all elements show up on the YZ Plane.

DXF Files that are not making use of the UCS Header variables are displayed correctly in my Program.

Autodesk:
[image]

Me:
[image]

Regards
Thomas

haplokuon, 2018-01-11:
All the coordinates of the entities are expressed either in world coordinates, global XY plane with normal (0,0,1); or object coordinates, which construction plane is defined by the normal of the entity.

The values of the header variables UcsOrg, UcsXDir, and UcsYDir are purely informative, they define the actual user coordinate system, but the stored coordinates of the entities are not affected by it. Any transformation from the user coordinate system to the world coordinate system, and vice versa, have to be done manually.

Daniel

Troubadix75, 2018-01-16:
Hello Daniel,

thank you for your explanation.

I thought the Lib would apply the UCS right away. But of course that is not possible.

closed 2018-01-17 by haplokuon