haplokuon/netDxf #20: How to set default ZOOM EXTENT?

WangJi, 2018-01-15:
Hi, I've successfully created a dxf file.
But the question is I can't find a way to set **view extent**, here is what I'm trying:

    var dxf = new DxfDocument();
    var layout = dxf.Layouts.First();
    layout.MinExtents = new Vector3(minX, minY, 0);
    layout.MaxExtents = new Vector3(maxX, maxY, 0);
    layout.MinLimit = new Vector2(minX, minY);
    layout.MaxLimit = new Vector2(maxX, maxY);
    dxf.Save(savePath);

However, nothing changes. Any help is appreciated.

WangJi, 2018-01-15:
Well, after a little digging, I found `dxf.Viewport.ViewCenter` is the key to view center.
Then where is the another `zoom level` key?
 `dxf.Viewport.ViewAspectRatio`  seems to take effect. Can anyone explain.

haplokuon, 2018-01-17:
I cannot help you about it, the dxf documentation leaves too many things unclear. Those are the variables, perhaps there are other, that supposedly control the view but I have not been able to find how exactly how they work. My fear, and this is pure expeculation I might be wrong, they are related with the view in the user interface, but the size of the window where the entities are displayed is not saved in the dxf.

The best way to deal with it is in your own code that shows the contents of the file. Once the file is loaded you can calculate the size of the projected vertices and scale the bounding rectangle to fit the window in you user interface.

Since I am mostly interested in the actual geometry I do not care too much about this. Any program, where you load the file, will have some kind command to make a zoom extent.

Daniel

WangJi, 2018-01-18:
That's totally cool.
Finally I manage to get things done by setting some magic number, haha.
Just as you said, we are only focused on generating vector file, for those who open it, why don't they just press `z  + e`.

closed 2018-01-18 by WangJi

WangJi, 2018-01-18:
By the way, I think I should let know that I'm working on a program that converts a `handwriting` image from raster to vector. The input picture is like this: 
[image]
And with the help of your awesome work, I now can easily export those converted results into `dxf` file like this:
[image]
Which saved me a lot of time at least a a week or more.