haplokuon/netDxf #21: when dxf code ($DWGCODEPAGE )is gb2312,load dxf doc will get exception

xiaohelong, 2018-01-16:
location:DxfReader.cs
      if (isUnicode)
                        encoding = Encoding.UTF8;
                    else
                    {
                        // if the file is not UTF-8 use the code page provided by the dxf file
                        if (string.IsNullOrEmpty(dwgcodepage))
                            encoding = Encoding.GetEncoding(Encoding.Default.WindowsCodePage); // use the default windows code page, if unable to read the code page header variable.
                        else
                        {
                            int codepage;
                            //encoding = Encoding.GetEncoding(!int.TryParse(dwgcodepage.Split('_')[1], out codepage) ? Encoding.Default.WindowsCodePage : codepage);
                            encoding = Encoding.GetEncoding("GB2312");// my code
                            //这里出错 索引超出了数组边界
                        }
                    }

xiaohelong, 2018-01-16:
//encoding = Encoding.GetEncoding(!int.TryParse(dwgcodepage.Split('_')[1], out codepage) ? Encoding.Default.WindowsCodePage : codepage);
encoding = Encoding.GetEncoding("GB2312");// my code

haplokuon, 2018-01-17:
Like with many other things the dxf documentation leaves too many things unclear, and this is one of them. The values of the $DWGCODEPAGE usually take the structure of ANSI_[codepage number], but this is only for the cases I have been able to test, always based in Latin characters, but for non Latin, like your case, it is a Chinese codepage, I do not know, they might be different. There are other Chinese users of this library but this is the first time someone has posted a problem with it.

What is the full content of the $DWGCODEPAGE header variable in the dxf file? Can you upload or post a sample file so I can take a look at it? Which is the software you used to generate the dxf?

For languages not based in Latin characters I would recommend to use Autocad2007 version or higher since they should use Unicode (UTF-8), but again I am not 100% sure, I cannot test what happens when you use, for example, AutoCad in Chinese.

Daniel

closed 2018-05-04 by haplokuon