haplokuon/netDxf #22: Invalid dxf

NapaAntti, 2018-01-26:
I have a dxf file which can be opened with acad, but the netDxf complains about invalid format at MLINESTYLE entry which seems valid according to dxf documentation.

MLINESTYLE
  5
9
330
8
100
AcDbMlineStyle
  2

 70
     0
  3

 62
   256
 51
90.0
 52
90.0
 71
     0
  0

I tried to provide the name at group code 2 but it did not make any difference. What could be the problem?

NapaAntti, 2018-01-26:
Seems that version 2.0.3. works if the name is provided to the group code 2, but the 2.0.1 version does not work

haplokuon, 2018-01-28:
The program that created that dxf should have provided a name for the MLineStyle, you cannot create one without a name, in the same way that a layer, a text style,.... all must have a valid name.

What happened with version 2.0.1, I don't remember, always use the latest version available to avoid problems.

Daniel

closed 2018-05-04 by haplokuon