haplokuon/netDxf #27: Feature request : Add $EXTMIN and $EXTMAX into Header

wangzhen-Geosurf, 2018-03-27:
Thank you very much for this wonderful library firstly.

Recently, I used this library in my project. It seems that the current version doesn't support the DXF header variables of **$EXTMIN** and **$EXTMAX** in **Header** section.
Can you add them in **Header** section?
And I attached the explanation of these two variables (Autocad 2012) for your reference as follows. 

[image]

Thanks in advance.

haplokuon, 2018-04-04:
I have not implemented those header variables, to properly do so it will require to calculate the bounding box of whole drawing. Reading them its easy but writing the correct ones will require to calculate the bounding box, it can be done but I have not implemented the code to do it. For some entities the calculation its pretty straightforward but for others like blocks will require a little more work.

The easiest way it is to leave the problem to the user and add a default one in case none are passed when writing the file. If this is enough for you it is not a problem to implement them.

Daniel

wangzhen-Geosurf, 2018-04-05:
@haplokuon 
Thank you very much for your reply.

What I want is as same as you said in the last sentence of your comment. I want the Header (HeaderVariable.cs and HeaderVariableCode.cs) to include these two variables with default values and `DxfDocument.Save()` function can export them into file. Absolutely, in this way, the user can calculate and set the values of them by themselves.

However, I'm worried about the display difference while opening the two kinds of exported DXF file (one: with default value of $EXTMIN and $EXTMAX; another: the current exported dxf file which is without of them) with some softwares like *autodesk trueview*.

haplokuon, 2018-04-16:
Added both variables as readonly. Since the library does not calculate them there is no need to write them for new dxf, so they will only contain valid values for loaded dxf that actually contains them, otherwise they will be set to Vector3.NaN.

In any case writing erroneous values for ExtMin and ExtMax should not harm the dxf in any way, my guess is they are always automatically generated whether they are present or not, at least it is what it seems with AutoCad.

Daniel

closed 2018-05-06 by haplokuon

patrickwensel, 2020-09-18:
Hi Guys!  Building a laser cutting estimating tool for a client.  I need to know how many parts will fit on on sheet of material.  I can't require the user to add the extents so I will need to calculate.  Any advice on how to get started?

haplokuon, 2020-09-22:
You can access those header variables through the CustomVariables collection in the DrawingVariables. Check the sample "AddHeaderVariable()" in the TestDxfDocument project.