haplokuon/netDxf #29: DxfDocument is null when loading a DXF

qprotex, 2018-04-23:
Attached there is a DXF file that is not loaded properly.

`DxfDocument dxfLoad = DxfDocument.Load("BICYCLE.dxf");`

That line does not throw any exception but dxfLoad is null.

[BICYCLE.zip](https://github.com/haplokuon/netDxf/files/1936280/BICYCLE.zip)

sorifiend, 2018-04-23:
The "Load" method will return null if the file has not been able to load.

Please verify that the "BICYCLE.dxf" in in the correct place and that your code can see it by using something like this first: `Console.WriteLine("File exists: " + File.Exists("BICYCLE.dxf"))`

sorifiend, 2018-04-23:
Ahh, I receive a `System.NotSupportedException`.

Your issue is that `Loading dxf files prior to AutoCad 2000 is not supported.`
I opened the file with CAD and resaved it as AutoCAD 2007 format, and the 'DxfDocument.Load()'worked perfectly.

qprotex, 2018-04-23:
Thanks. Any chance the Load method can throw System.NotSupportedException in future versions?

sorifiend, 2018-04-23:
That is a question best asked of the developer haplokuon, but I imagine that there are so many things that can cause the file not to load, and there is very little way to tell between something being unsupported verses something being corrupt etc.

You are welcome to investigate and submit a pull request to add the feature, but its easy enough to catch a generic exception or show a generic error to a user whenever the return is null, and just state that it is an unsupported file, and that they should manually resave the file in a newer format (AutoCAD 2007 in my case).

Source of the dxf loading (`DxfDocument Read(Stream stream)`) is here if you want to take a look: https://github.com/haplokuon/netDxf/blob/master/netDxf/IO/DxfReader.cs

haplokuon, 2018-04-23:
There are two methods in the class DxfDocument exactly for this purpose, called:
```c#
public static DxfVersion CheckDxfFileVersion(Stream stream, out bool isBinary)
```
and
```c#
public static DxfVersion CheckDxfFileVersion(string file, out bool isBinary)
```
In your case it will return AutoCad14. netDxf is compatible with dxf versions AutoCad2000 and upwards, anything lower and it will not be able to read it. You can use the Debug build if you want the library to throw any exception that might occur during the reading process. The thing is that, if the file is one of the compatible dxf versions, and the Load method still returns null there is nothing you can do. It might be a bad formatted file or a bug, in that case I will need to take a look at it to find a fix.

Daniel

v-tsymbalistyi, 2021-08-12:
And what if CheckDxfFileVersion returns AutoCad2000 and I am still getting null as a result of DxfDocument.Load?

LeaderXinjianSun, 2022-02-18:
You can copy all elements to to a compatible dxf file which you can load successfully, then save as it.The new file can be loaded successfully too.

DmitryDronovDre, 2022-08-23:
> You can copy all elements to to a compatible dxf file which you can load successfully, then save as it.The new file can be loaded successfully too.

Thx, it work for me too. File that was saved from autocad was not opened. but show version DXF. When i save els at new file - document start to reading. Thx.

haplokuon, 2022-08-26:
@DmitryDronovDre

I got your file " MKM_HdM_EP_100_GR_G00_x_x_01_Grundriss.dxf" and you need to update the library using the latest source code. It seems that the problem you have loading the file is related with a bug while reading the layer states, that has already been fixed.

Regarding the reduction in file size that may happen. Overtime the drawing can accumulate a lot of stuff that is not in use anymore like blocks, styles, layers,... If you use AutoCad that exactly the purpose of the "purge" command clean the drawing of unreferenced objects.

yoonsungkim76, 2022-12-07:
Hi haplokuon.
Here is a sample dxf file(cad 2000) still have null issue. could you check this file?
[test.zip](https://github.com/haplokuon/netDxf/files/10172695/test.zip)

haplokuon, 2022-12-08:
@yoonsungkim76
If you have a new problem open a new Issue, do not hijack unrelated old posts.

First make sure you use the latest available source code. I found a small problem with your DXF, there are several 1071 codes out of place in the Layer data. If you use the Debug build an assertion (not an exception) will be raised showing that there is something abnormal with your file but will allow you to continue loading it. If you use the Release build the problematic lines will be skipped and the loading process will continue normally without raising any exception.