haplokuon/netDxf #34: How to stretch an entity

zgyijg, 2018-05-04:
How to draw a box or a cylinder?

zgyijg, 2018-05-04:
How can I do？thank you

haplokuon, 2018-05-04:
Which is your question, how to stretch an entity? or how to draw a box or a cylinder?, or both. Anyways, in both cases you will need to do it manually, netDxf does not pretend to solve geometric problems but help you reading and writing dxf files. You will not find any kind of entity that is not directly supported by the dxf specification.

In the case of drawing a box or a cylinder. First you will have to decide which dxf entity you will want to use, a bunch of Face3d, a PolyfaceMesh, or a Mesh; and that will depend on what you want to do with it. Second build its vertices and its faces according to the chosen entity, how to build the data is pretty straight forward and you can find lots of samples searching a little bit.

Daniel