haplokuon/netDxf #36: Dxf to Pdf

tyrgin, 2018-05-10:
May I ask if you can suggest me any dxf to pdf converter for printing tonns of drawings?
I'm not sure, you don't have it in netDxf?
Thank you in advance

waryhermit, 2018-05-15:
netDXF does not support any kind of rendering. So yes, it cannot export to pdf. I do not know a good free API for that, please post here if you do find one.

haplokuon, 2018-05-15:
I cannot recommend a pdf converter, I have not used any. In the occasions I needed a copy of a drawing in pdf, I just printed it using one of the available virtual pdf printers, like PDFCreator or Bullzip.

Daniel

waryhermit, 2018-05-15:
I can get the dxf's to PDF manually OK. I just want to make generating a pdf part of the automated drawing process I have made with .netdxf.

tyrgin, 2018-05-16:
I use AutoCad console mode for pdf, works not very fast but still 2k prints in hour

waryhermit, 2018-05-16:
Hmm I only have Autocad LT 2012. Do you know if that is available in my version? Do you know of any free product that could do it via command line / api?

Also 2k prints an hour?? What kind of company generates drawings at that rate :scream: 
I doubt we will ever get to more than 100 or so an hour, if that.

tyrgin, 2018-05-16:
Nah, Autocad concole is available after 2013v. 

I did not say we create new docs with that speed, but we update them.
We have about 60k documents to rework but unfortunately not all of them in dxf( some part in dwg

haplokuon, 2018-06-06:
Did you try [QCAD](https://qcad.org/en/)? If you work only with 2D drawings, the professional version is quite cheap and I believe it will allow you to convert drawings to pdf from the command line.

Daniel