haplokuon/netDxf #40: Dimension Style Dataloss

waryhermit, 2018-05-15:
Steps to reproduce:

- Create and save blank dxf file with netDxf
- Open with autocad LT or draft sight
- Add a new Dimension Style - set Dimension text placement to centered for both vertical and horizontal
- Set linear dimension precision to four decimal places
- Save as dxf
- Reopen and save using netDxf
- Open file in autocad lt / draftsight
- Check you Dimension style - the name will be the same but the centering and precision values are lost and defaulted.

haplokuon, 2018-05-15:
Please attach your sample dxf, so we will be sure we are working with exactly the same file.

Daniel

waryhermit, 2018-05-15:
I have attached a zip file with example dxf files.

**Blank.dxf** is the blank dxf file generated with netdxf
**BlankWithStyle.dxf** is the same file, but after I have opened it in Autocad LT 2012 and added a dimension style, TestStyle
**BlankWithStyleAfterResave.dxf** is the above file, after opening and re-saving with netdxf

[ExampleDxfFiles.zip](https://github.com/haplokuon/netDxf/files/2006208/ExampleDxfFiles.zip)

waryhermit, 2018-05-22:
Is there anything I can do to help you figure this out / fix it?

haplokuon, 2018-05-23:
I had enough information, the dxf output from the program you were using it's all I needed, keep in mind that I might or might not use the same software and the output also might vary from program to program. I was just busy with other matters.

About the number of decimal places the dimension style in your file was not saving the code for that value, therefore the default was used instead, netDxf had the default set to 2 instead of 4. Something similar with the text placement, netDxf default is vertical "above" and horizontal "centered". I guess they have changed the default values over the years or I had them wrong from the beginning, one way or the other I will change them. The official dxf documentation it's not very helpful with this matter, it does not specify in many cases which are the default values.

One important thing about the dimension style text placement is that netDxf only supports Above-Centered when generating the block that represents the dimension, I cannot measure text strings. At the moment, you cannot manually change those values but the original values in the dxf should be preserved.

I might allow to set the text placement with a big advise that only Above-Centered is supported, so you can modify those values to the supported ones when loading dxf files into netDxf. I will need to think about it. The dimension style has lots of variables and not all of them are saved into the DimensionStyle class, so similar problems might arise with the not common values.

Daniel

haplokuon, 2018-06-02:
It should be fixed now with latest commit netDxf 2.1

Daniel

closed 2018-06-06 by haplokuon

waryhermit, 2018-06-07:
I applied this update and now I am getting the error "Only true type TTF fonts and acad compiled shape SHX fonts are allowed."  Is that expected?

waryhermit, 2018-06-07:
Apperently there was some references to a font called txt / TXT - I opened the template dxf file I was using with a text editor and replaced the references to that font with something else and it worked.

haplokuon, 2018-06-08:
I guess this time the problem lies in the program that generated the DXF. From what I found out, this is speculative the DXF documentation sucks, the TextStyle font is defined by the font file or when there is none the font should be defined in the extended data.

This is something the AutoCad documentation says about true type fonts "TrueType (TTF) fonts must be installed by the operating system so that they are listed in the system registry." SHX, compiled shapes fonts, are the other kind available in AutoCad; so, yes that error is expected to avoid further problems when importing the DXF somewhere else. Who knows what TXT is, it is better to not write something rather than something wrong.  

In any case this is a problem easily avoidable I will just give it a default font when there is a problem with it.

Daniel

waryhermit, 2018-06-08:
@haplokuon Yeah, defaulting to a certain font or just give the font name in the exception or something so its easier to figure out what it going on. That is how I figured it out, I just added the font name to the exception you already had there which lead me to the strange "txt" font.