haplokuon/netDxf #41: Using the library in Unity2017

guozi110001, 2018-05-25:
I have an issue for using the library in Unity2017,when I new DxfDocument have an error "Failure has occurred while loading a type".I thought the Unity probably use .net framework3.5 and netdxf use 4.5,so I convert netdxf to .net framework 3.5, but it does not work.
Has any one have a way to do this?
here is the error report:
```
TypeLoadException: Failure has occurred while loading a type.
netDxf.DxfDocument.AddDefaultObjects () (at <703e308ee7344ce798f3d765ee824ef9>:0)
netDxf.DxfDocument..ctor (netDxf.Header.HeaderVariables drawingVariables, System.Boolean createDefaultObjects) (at <703e308ee7344ce798f3d765ee824ef9>:0)
netDxf.DxfDocument..ctor (netDxf.Header.HeaderVariables drawingVariables) (at <703e308ee7344ce798f3d765ee824ef9>:0)
netDxf.DxfDocument..ctor () (at <703e308ee7344ce798f3d765ee824ef9>:0)
cadTest.LinearDimensionTest () (at Assets/Scripts/cadTest.cs:497)
cadTest.Update () (at Assets/Scripts/cadTest.cs:34)
```

leothyssen, 2018-07-05:
I think i made it to work!
I updated to unity 2018.1.4f1 and, in the build settings-player settings,  changed the "Scripting Runtime Version" to ".NET 4.x Equivalent". 
Then i compiled the standalone netdxf project and copied the netDxf.dll and netDxf.pdb into Assets/Plugins.

I can run the sample code from my unity project by adding
"using netDxf;
using netDxf.Entities;
using netDxf.Header;"

It works just fine, and creates a "sample.dxf" file in my unity-projects folder. 
However, the created file crashes all of my viewers, despite the fact that it looks like a legit dxf file when i peer into it with a texteditor.
I'll see how far i get from here.

haplokuon, 2018-07-09:
From what I know (I do not use Unity myself) it seems that Unity 2018 uses the net Framework 4.6 now. So, I guess it should be no problems using libraries that reference to earlier builds of the net framework, but in case it does, keep in mind that this is purely speculative, one thing you can try is to download the netDxf source code and change the net framework of the library to the same one that Unity uses. There should be no problems going up from 4.5 to 4.6, the problem is going down.

@leothyssen If need it I can take a look at your file.

Daniel

leothyssen, 2018-07-09:
Sorry my bad! I tried to isolate the error and opening the file([sample.dxf.zip](https://github.com/haplokuon/netDxf/files/2176573/sample.dxf.zip) ) with [LibreCAD](https://sourceforge.net/projects/librecad/?source=typ_redirect) and [FreeCAD](https://www.freecadweb.org/wiki/Download) works fine. 
Thanks for the replies, this is an amazing(!!!) project! I'll see how far i can push unity with it. 

To refer to the .net issue: There was no problem for me and my unity version (2018.1.4f, .NET 4.x Equivalent).
Just when opening it with the Autodesk online viewer (and some other online viewers) it throws:
> AutoCAD-InvalidFile Sorry, the drawing file is invalid and cannot be viewed. - Please try to recover the file in AutoCAD, and upload it again to view.
> TranslationWorker-InternalFailure Unrecoverable exit code from extractor: -1073741831

But as long as it works with the standalone cad viewers (even though [this one ](http://www.edrawingsviewer.com/ed/download.htm)didn't work) that's good enough.

leothyssen, 2018-07-09:
oh, but opening it with [SPIRIT](https://www.softtech.de/en/products/spirit) (version 2015.4.2015.714) throws an error 
> EExternal Exception E06D7363

as well, is that a common issue with proprietary dxf viewers?

leothyssen, 2018-07-10:
Hey Daniel, 
It worked! The AutoCAD online viewer takes all versions >= AutoCAD2007! That's great, thanks a ton for the hint. But I'm surprised by your remarks on the non-latin encoding, 
-as I'm coding on a MacBook Air(High Sierra, Keyboard Layout: German, System Language: English), and 
-as the header section with 

>3
>ANSI_1200

is the same in all 5 Versions. 

Leo

[image]

ecarrera, 2019-05-10:
Awesome library @haplokuon, I have implemented it to show the drawings in Unity from downloaded .fbx of server.

Thanks @leothyssen for the steps to start with Unity, it was pretty useful for me.

Both of you have made my day.

I was wondering why there is no support for LT versions? Do an update will deal with it? What about further versions of Autocad?

haplokuon, 2019-05-13:
Are you sure you posted in the right project? Your comment has left me a little bit confused, this library is to load and save DXF not FBX files.

On the other hand, AutoCad LT versions are supported in the same way as any other program that works with DXF files. And for future versions if a new DXF database version is created, most probably, the only thing required will be to add a new entry to the DxfVersion enum. For what concerns to netDxf nothing has changed since 2007.

ecarrera, 2019-05-13:
> Are you sure you posted in the right project? Your comment has left me a little bit confused, this library is to load and save DXF not FBX files.
>
> On the other hand, AutoCad LT versions are [...]

Ohh sorry, I meant "downloaded .dxf files from server". **_Comment edited._**

Yeap, about Autocad LT versions I see that versions < 5 are not supported:

DxfReader.cs
Lines: 140, 141
```
if(version<DxfVersion.AutoCad2000)
    throw new DxfVersionNotSupportedException(string.Format("DXF file version not supported : {0}.", version), version);
```

DxfVersion.cs
Line: 63, 64
```
[StringValue("AC1015")]
AutoCad2000 = 5,
```

So, my question is, why there is no support for these versions (AutoCad10, AutoCad12, AutoCad13, AutoCad14).

haplokuon, 2019-05-16:
Without going into many details, there are some important changes where introduced with the AutoCad2000 DXF version, and they are so old that I never care about them (in AutoCad all those old versions are saved as AutoCad12).

Any program that I know that works with DXF files are capable of saving to more updated versions. I would even recommend to use the AutoCad2007, and later versions, since they are saved as Unicode. 

This is applicable not only to AutoCad LT but all programs that saves DXF files. The nomenclature I am using is the one that appears in the DXF official [documentation](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-A85E8E67-27CD-4C59-BE61-4DC9FADBE74A). It is the $ACADVER header variable.

ecarrera, 2019-05-17:
Thanks for the explanation @haplokuon . I was dealing with old .dxf files and it gave me the errors, this is why I had to ask you about it. In effect I'll start using AutoCad2007 and later. Thank you very much for this library. 👏

13704017891, 2019-06-03:
@leothyssen   Hello, excuse me, how to use this library in unity?I am doing the same thing. Can you refer to your project?

ecarrera, 2019-06-04:
> @leothyssen Hello, excuse me, how to use this library in unity?I am doing the same thing. Can you refer to your project?

You have to build the library and copy the .dll to the "plugins" folder in your Unity project.

IronGamesCom, 2019-06-08:
@leothyssen Hello, excuse me, how to show drawings in Unity with this library?   I have implemented it to load, edit and save my dxf, but how to show it in Unity?

HzTom111, 2019-06-10:
Hellow

roometric, 2019-06-10:
Hey, sorry i have only generated the dxf files from gameobject data in unity, never vice versa....

leothyssen, 2019-06-10:
Hi, sorry i have only used the plugin to export dxf files from gameobject data present in unity, never vice versa... should be possible though somehow, let me know if you found a solution: it would definitely interest me. But there should definitely be some other unity-native dxf viewer plugins. Maybe you can use one of them to display the files, and this library to create them. Just search on the assetstore.

ecarrera, 2019-06-10:
Using .dxf to create Unity gameObjects you have to create them like primary elements, a .dxf file contains a lot of data but main used in Unity may be faces3d, polylines and lwpolylines.

With faces3d you can use the vertices and triangles to create a mesh, but you have to know how to get triangles and its vertices from a face3d.

Polylines and LWPolylines can be rendered with LineRenderers getting its coordinates and colors, then setting the points for the LineRenderer.

As LWPolylines are 2D polylines (no deep), you can create it in a canvas, if preferred.

Here you can get the basics about constructing a Mesh, [](https://ilkinulas.github.io/development/unity/2016/04/30/cube-mesh-in-unity3d.html)

And about a basic knowledge there are the Unity concepts [](https://docs.unity3d.com/Manual/AnatomyofaMesh.html)

ecarrera, 2019-06-10:
@IronGamesCom @leothyssen 

Using .dxf to create Unity gameObjects you have to create them like primary elements, a .dxf file contains a lot of data but main used in Unity may be faces3d, polylines and lwpolylines.

With faces3d you can use the vertices and triangles to create a mesh, but you have to know how to get triangles and its vertices from a face3d.

Polylines and LWPolylines can be rendered with LineRenderers getting its coordinates and colors, then setting the points for the LineRenderer.

As LWPolylines are 2D polylines (no deep), you can create it in a canvas, if preferred.

Here you can get the basics about constructing a Mesh, [https://ilkinulas.github.io/development/unity/2016/04/30/cube-mesh-in-unity3d.html](https://ilkinulas.github.io/development/unity/2016/04/30/cube-mesh-in-unity3d.html)

And about a basic knowledge there are the Unity concepts, [https://docs.unity3d.com/Manual/AnatomyofaMesh.html](https://docs.unity3d.com/Manual/AnatomyofaMesh.html)

13704017891, 2019-06-11:
@ecarrera  Hi, I already know how to use this library.I have also read out the data, but I don't know how to use the data to render them in the unity scene, can you give me some solutions?I am confused now. My email is 13704017891@163.com.Thank you very much for your answer

13704017891, 2019-06-11:
@ecarrera Hi, I already know how to use this library.I have also read out the data, but I don't know how to use the data to render them in the unity scene, can you give me some solutions?I am confused now. My email is 13704017891@163.com.        Thank you very much for your answer

13704017891, 2019-06-11:
@ecarrera  Is it possible to draw a DXF model in OpenGL?

ecarrera, 2019-06-11:
> @ecarrera Is it possible to draw a DXF model in OpenGL?

I have not done it before but look to be easy with the data segmented in the DXFDocument, maybe is the same way as I do in Unity for my cases; rendering 3d models (MeshRenderer), 2d lines (LWPolylines) and 3D lines (polylines) with LineRenderers.

If you need to render a 3D model from the faces3d of the DXFDocument here is a starting guide: [https://learnopengl.com/Model-Loading/Mesh](https://learnopengl.com/Model-Loading/Mesh)

And for LWPolylines/Polylines here: [https://artgrammer.blogspot.com/2011/05/drawing-nearly-perfect-2d-line-segments.html](https://artgrammer.blogspot.com/2011/05/drawing-nearly-perfect-2d-line-segments.html)

13704017891, 2019-06-12:
@ecarrera Thank you so much for your reply and advice

marioslokas-v, 2020-12-10:
Hey, sorry for necro posting, but have people been able to write mesh data from Unity directly to DXF with this library?

ecarrera, 2020-12-10:
@marioslokas-v the lib help you to parse DXF to C# object, having a DXF object in Unity then you can create meshes, polylines 2d, 3d, etc etc. Also it is better than other libs, It is really quick to parse big amount of data and to create meshes with millions of triangles in seconds.

marioslokas-v, 2020-12-10:
Thanks for the reply. Do you happen to know if the lib is also using threads or anything that can't be used in a Unity WebGL build (advanced IL2CPP features)? My goal is to export an DXF file in runtime from user-created gameobjects.

ecarrera, 2020-12-10:
It only parse a dxf to a c# object, I guess it doesn't need threads. I remember I made a built for WebGL.

marioslokas-v, 2020-12-15:
After a bit of work it has succeeded to convert a mesh to DXF. Quite straightforward, just had to make different entities for child transforms with a mesh renderer. Also working from a WebGL build. Thanks a lot for the support.