haplokuon/netDxf #42: Insert Exploding

philip556677, 2018-06-07:
I see where the code was planned on being included but when uncommented and used, the results are incorrect.
Is there any method implemented that will allow for the exploding of inserts?
Or any idea of an equation that would properly position each item in the insert?

haplokuon, 2018-06-08:
I guess you are talking about the commented code in the Insert class. I wrote it long time ago and something might be broken, perhaps I will continue with it in the future, even if, at the beginning, the full set of entities is not available.

There is no magic equation, just some matrix and vector operations, that will allow to explode correctly all available entities, some of them even require special treatment.

Daniel

haplokuon, 2018-06-13:
That explode code is very old and required some changes, the modifications you have made should make it work once again. I renamed the property of Block Position to Origin, and also it was required to consider the units of the document and the block to properly apply the scale. I guess I modified those things after that explode code, I really don't remember.

Maybe I will pickup that code even thought at the beginning, the insert's explode method might not be able to explode the full set of available entities.

Daniel

philip556677, 2018-06-13:
I was able to get the code working for lines and text.
I'll go ahead and work on the rest of the entities and provide you with the code if you want to implement it.

haplokuon, 2018-06-15:
Ok, go ahead, sometimes I forget things I was doing until someone comes and ask about them.

Daniel

Wzshun, 2018-08-23:
@haplokuon 
I want to know the affect order for Insert 's base position, scale and rotation when them are applied to affine matrix.
When deal with exploding Insert, is it some types of entity will be out of the shape? such as Circle scale to ellipse?
I find that if i want to draw all entities in DXF file, it has to deal with Insert exploding.

haplokuon, 2018-08-24:
The order of transformation are scale (insert's Scale method), rotation around the normal axis (insert's Rotation method), rotation by the normal axis (insert's Normal method), and translation (insert's Position method) that should also include the Block origin that must be transformed by the transformation matrix (scale, rotation, normal) since it is defined in the Insert's local coordinates, I always recommend to keep this value as the zero vector it only adds confusion without any real purpose. Keep in mind that I use column vectors (column major representation) therefore matrix multiplication is performed from right to left (A·B)t = Bt·At, where A and B are matrices and t indicates the transpose matrix.

Yes, the Circle and Arc are the only entities that, in the case of a non-uniform scale, get transformed to another type of entity, an Ellipse. An ellipse can become a circle or arc but since it a circle can be interpreted as an ellipse with equal axis changing its type is not necessary.

archgame, 2022-02-23:
@haplokuon I'm having trouble with Insert Exploding ([following this link about iterating nested blocks](https://github.com/haplokuon/netDxf/issues/133)), when I explode my Insert, it only returns 159 Line objects, when there should be over 1000 objects including 10 MText objects, has @unitedsoftwork 's code been implemented into the library? I couldn't find the [Explode method in the Insert class](https://github.com/mfarquhar/netDXF/blob/master/netDxf/Entities/Insert.cs)?

haplokuon, 2022-02-27:
Without more information or the original DXF file I have no idea what might be happening.