haplokuon/netDxf #43: Page setup data not preserved (PLOTSETTINGS?)

waryhermit, 2018-06-07:
Page setup of template file before saving with netDXF:

[image]

Page setup after saving with .netDxf:
[image]

In the zip file there are two files:

- test.dxf - This is the template file after my program has used .netDxf to generate the model. The plotting information seems gone.
- testWithPageSetup.dxf - This is the file after I have gone back and added the page setup again (Plot extents, fit to page.)

[Output Drawings.zip](https://github.com/haplokuon/netDxf/files/2081869/Output.Drawings.zip)

haplokuon, 2018-06-08:
There are some properties missing in the PlotSettings class, nothing serious, pretty straightforward. I will add them in next update.

Daniel

waryhermit, 2018-06-08:
Excellent, thank you.

haplokuon, 2018-06-15:
All PlotSettings parameters should be now available.

Daniel

closed 2018-06-18 by haplokuon