haplokuon/netDxf #44: a new cad/cam app for cabinet doors

akdeniz23, 2018-06-08:
Hi Daniel,  here what we discussed previously, thnak you for your answer.
-----------------------------------------------------
Here goes a few tips. From your images it seems that you only need to work in 2D with very basic entities like lines, arcs, and polylines; splines perhaps?. I will not recommend you to build your own CAD application too much effort, it is not an easy task, when there is something that can suit your needs, like QCAD, it even has a version focused on CAM to generate G-Code files. The software is very basic but it is cheap too. You can also try the free version of DraftSight.

What I would do is build an application to generate parametrically the drawings, netDxf can help you with that generating DXF files that you can open later in QCAD, kinda like the dynamic blocks in AutoCAD. I am assuming that you might work with a set of designs that can be parameterized for door sizes, margins, arc radius,...; for other designs is no problem drawing them directly with QCAD.
------------------------------
Now my question is ,I am still undcided which CAD app to use to draw kitchen cabinet doors for CNC machining?  I am leaning towards CMS IntelliCAD because it is based on IntelliCAD which is a very good CAD app after AutoCAD itself. I can save the line, spline paths separately from dwg, stl etc. as my own  data.
Then use this data to construct a toolpath for GCode.  I need to check QCAD as you said but CMS IntelliCAD is hard to beat as an AutoCAD clone. What do you think? thnak you

akdeniz23, 2018-06-08:
QCAD is 2D, I may need 3D for visualizing the drawing now. But in the future I will definetely need 3D. thnaks

haplokuon, 2018-06-09:
Well, you ask for free or cheap CAD solutions. My last experience with IntelliCAD was a very long time ago, and I am not sure if they, have a free version. At the end of the day use the software you feel more comfortable with. I have no experience with cad software specialized for CNC machines, in the past, I had to prepare drawings for them, but I never had to work with them directly.

I will only recommend free software. For paid solutions, I will have to start to ask for a fee to the company you buy the software ;) Sometimes people has the presumption that because it is free software it cannot be good, when many times, for most people, is good enough. Take, for example, Microsoft Office, people is obsessed with it, when you have LibreOffice, how many times did I hear "but is compatible with doc? yes, with doc, docx, and odt, the document format that you should be using, in the first place, instead of doc. I am not talking about software solutions for big buck companies, that is a whole different story.

But as I said you do not need to draw a single line, correct me if I am wrong, but I supposed, most of the time, if not always, you start with a set of predefined designs that you can easily build them programatically: the drawing plans, the paths for CNC machines, even 3d models.

Good 3D CAD/CAM/CAE software is expensive, but if you require 3d software for visualization purposes use [Blender](https://www.blender.org/), nothing beats it, it is an amazing piece of software that is free and open source. Check its archimesh plugin,... it will show you how to parametrically build 3d models focused on architectural stuff, it has kitchen cabinets.

Daniel