haplokuon/netDxf #45: Question : block rotation

Pixelstudio, 2018-06-12:
Hi there,

I'v been searching all over but did not find something to help me out.

i load a block and place that on specific points, something like this :
` 
  var newBLock =(netDxf.Blocks.Block) block.Clone();
  Insert nestedInsert = new Insert(newBLock) 
		{
			Position = position,
			Normal = normal,
			Rotation = rotation
		};
		nestedInsert.TransformAttributes();
        document.AddEntity(nestedInsert);
`

This way i specify an normal and a rotation value.
However i would like to rotate the inserted block over another axis, how can this be done? 
i can only rotate an block one axis right ?

Also why is the rotation not a property (like position and scale) ?

Best regards,

haplokuon, 2018-06-13:
The normal and rotation represents the [axis-angle representation](https://en.wikipedia.org/wiki/Axis-angle_representation) of a rotation, to properly define the orientation of your insert you have to work with both parameters at the same time.

The insert Rotation IS a property, you could no initialize the Insert the way you do if it were not one.

Daniel

Pixelstudio, 2018-06-14:
okey, let me reprase.

Let say i have inserted my block :
`Insert nestedInsert = new Insert(newBLock) { 
Position = position, 
Normal = normal, 
Rotation = rotation }; 

nestedInsert.TransformAttributes(); 
document.AddEntity(nestedInsert);`

i have rotated the block around the normal for x degrees, not i would like toadditional rotate the object (on another normal axis).

how can this be done ?

haplokuon, 2018-06-15:
I would use quaternions. The [OpenTK](https://github.com/opentk/opentk) project has a nice [quaternion](https://github.com/opentk/opentk/blob/develop/src/OpenTK/Math/Quaternion.cs) class that will help you to operate with them and to convert between different rotation representations like rotation matrix, Euler angles (pitch, yaw, roll), and axis-angle.

There are lots of documentation about quaternions everywhere.

Daniel