haplokuon/netDxf #46: Error in netDxf.Entities.line

RamonAguila, 2018-06-15:
I have a project creating a dxf and writing something in it. In my computer, where I have installed visual studio, the project runs Ok, even running the installed .exe. But when I try to run the EXE in other computers it hangs giving my an error in "For each line as netDxf.Entities.Line in dxf.Lines" . Any clue what I´m doing wrong?

This is part of my code...

`Dim dxf as New netdxf.dxfdocument
Dim n as integer=0
Dxf= DxfDocument.Load(Me.OutfileDXF)
For each line as netDxf.Entities.Line in dxf.Lines
…..
`

haplokuon, 2018-06-15:
Does your other computer have the NET Framework 4.5 installed? Without more information I have no idea what might be happening. The code you are showing does not show nothing wrong apparently, the only two things I see, but they not should not create problems in VB are:
- There is no need to initialize the dxf variable with the "New" keyword, the static Load method will do it.
- Be careful with the casing, while VB is case insensitive other languages are not, in C# for example, dxf and Dxf are two different variables.

Daniel

RamonAguila, 2018-07-13:
I found the code is only working in Debug mode with Visual Studio, when I install the addin I made the error rises, it´s very strange cos the library is opening the file and it even checks version, but when i try "netdxf.entities,line" it breaks; the dxf´s have a "line" section, so I don´t have any clue about the problem.

haplokuon, 2018-07-14:
I have no idea, I do not even understand what you are trying to explain. You have two ways to post a problem, bug, issue, whatever. If a DXF file is giving you problems the best way is to upload it so I can take a look; if your problem is with the code, try to isolate the problem and post a full example that I can reproduce, if I cannot reproduce it it is very hard for me to give you an answer. If you need to write 50 lines do so, 3 lines only gives you space to explain the most basic of problems.

The first thing you should do is to start fresh, build a console application, add a reference to the netDxf.dll, and write a piece of code that shows your problem.

Daniel