haplokuon/netDxf #47: Need netDxf2013.sln to compile in Visual Studio 2013

2601krish, 2018-06-25:


2601krish, 2018-06-25:
I have Visual Studio 2013, so i could not compile netDxf project, it started throwing "the name 'nameof' does not exist in the current context" errors.
I need to draw a very simple basic shapes like line, circle, arc, rectangle, bezier curve etc.. with dimension texts and have the same in DXF file format. I Mean, my DXF file is a simple outline of some objects.
If you could provide netDxf2010.sln also, It will be very useful.
Thanks,
Krishna

haplokuon, 2018-06-26:
The problem is not with the Visual Studio solution file but with the C# version. The nameof keyword, between other things, is a new fetature of C# version 6.0 that was introduced in Visual Studio 2015. The C# and the Visual Studio versions are two different things,

If you work with a other VS version or use a different IDE, you can download the compiled binaries from [here](https://github.com/haplokuon/netDxf/releases).

Daniel