haplokuon/netDxf #49: why table is not support? I mean the entity like excel table

ytukids, 2018-07-10:
hi ,thanks for your work that i can  use netdxf conveniently.but i what know it is not support the table entity  which is not tableobject but like excel table

haplokuon, 2018-07-10:
The Table entity is not implemented because it is not, sorry that it sounds like a nonsensical tautology. If you look the Table entity documentation there are 6 pages and a half of codes. Reorganizing and giving some sense to them takes a lot of time. Something I will not do is just create a container with all the codes, and let you figure out what is the purpose and incompatibilities of each one. And the case of the MLeader entity (multi leader) is even worse. In the projects front page you can see the full list of available entities.

At least as @sorifiend has shown, you can access the information contained in the tables, you cannot create them but you can access to them.

Daniel

fjod, 2021-07-02:
Is there a way to edit info in tables? I can parse table contents; but when I edit and save doc, changes are not saved. I can put edited table in new document though.