haplokuon/netDxf #51: Getting Error in few types of files of hatches, while creating autocad files, in vb.net

riteshkhadka, 2018-07-24:
I am using this this for writing dfx file using VB.NET
Error: Object reference not set to an instance of an object

`Dim boundary As List(Of HatchBoundaryPath) = New List(Of HatchBoundaryPath)() From {New HatchBoundaryPath(New List(Of EntityObject)() From {poly})}
 Dim hatch As Hatch = New Hatch(HatchPattern.FromFile(Application.StartupPath & "\hatches.pat", project.Layers(i).PolyLine(k).HatchPatName), boundary)
  hatch.Pattern.Angle = project.Layers(i).PolyLine(k).HatchAngle
  hatch.Pattern.Scale = project.Layers(i).PolyLine(k).HatchScale`

Throwing exception in Dim hatch As Hatch. Some hatches name are shown but few are not like
"ISO11W100"
"ISO12W100"
"ISO09W100"
"ISO13W100"

haplokuon, 2018-07-30:
First, and this really goes to everybody, next time take your time and properly format your post. It will only take ten minutes to master the basics of the markdown formatting syntax. Also, if you are posting a piece of code make sure that I can make use of it, isolate your issue and build the most possible sample that I can run to be able to reproduce your problem.

Having said that, there is a problem with the PAT file parser in the HatchPattern.FromFile method that will fail to load the pattern definition under specific circumstances, but it does not generate the error "Object reference not set to an instance of an object" that seems you have seen.

Daniel

riteshkhadka, 2018-07-30:
It was not a issue. Since I was looking into the codes other have done and pretty new to VB. Finally, I got the solution, the hatch name that was being sent was not matching in the .pat files. So it was not a problem. It was hatches name mismatching. Thank you

closed 2018-07-30 by riteshkhadka