haplokuon/netDxf #52: Update nuget package

twilightrus, 2018-07-30:
Please, update nuget package. I have a some bug (in reader DXF-files) in your library from nuget (last updated 2 months ago), but if i build project from source code and add this as reference - this bug is fixed.
I think nuget package want updated since a c202cf91fcea3b3ae0b2a713854b11bdc562fe95 commit.

haplokuon, 2018-07-30:
You will have to contact the actual owner-maintainer of the Nuget package, I am not. Everything I provide is what you can find here at Github.

Daniel

zoran-petrovic-87, 2018-08-03:
I have the same problem with ver. 2.1.0. When loading from file it always returns null. I contacted Nuget owner, I hope that he/she will update it soon and that this problem is solved in 2.2.0.

BTW, Daniel excellent library, thank you!

Thieum, 2018-08-03:
2.2.0 has been uploaded - https://www.nuget.org/packages/netDXF/2.2.0

closed 2018-08-03 by twilightrus