haplokuon/netDxf #55: Is it possible to create associated dimensions?

waryhermit, 2018-08-16:
Is it possible to create dimensions that follow changes to the model? Right now I am manually creating dimensions in paperspace, and then overriding the measurement value so that it matches the model length. This doesn't seem quite "right" so I was wondering if I am missing anything.

haplokuon, 2018-08-20:
No, associative dimensions are not available.

ky3hec, 2019-07-29:
Hi! DXF specification says that **DIMASSOC** object should be used to create associatin between a dimension and geometry object. Have you already implemented this object? Any plans to do that?
Is there any other way to create association between dimension and geometry object?

haplokuon, 2019-08-06:
At the moment you cannot create associated dimensions, like with many other things there is no information in the documentation about this and the only way to figure it out how it works is looking into the DXF file and try to deduct how they are saved, and this is sometimes easy and sometimes isn't. This is something I would like to do, but I cannot tell you when or if I will.