haplokuon/netDxf #56: Is it possible to compile netDxf with .Net4.0?

lizhongchao13, 2018-08-22:
IReadOnlyList<T> dose not supported by .net4.0

haplokuon, 2018-08-24:
Yes, but you will have to manually solve any incompatibilities. If your only problem is the IReadOnlyList interface (I have not tried it) you can substitute it with something else. The IReadOnlyList is used where the access to the items in the list is needed but you cannot add or remove items from it.

As a general rule you can substitute private fields using the IReadOnlyList interface with a List and the properties that give you access to this value by a ReadOnlyCollection using the AsReadOnly() method of the list class. In the special case of the AttributeCollection that implements the IReadOnlyList, you can change it by an IList and throw an exception where add or remove operations are performed.

If you just substitute everything with a List you will have to remember that you cannot add or remove items from them or you may find many kinds of problems.

pucx, 2018-10-15:
enum.hasflag is not  supported too.

lizhongchao13, 2018-10-15:
thanks，i had compile it succeeded with net4.0

On 10/15/2018 03:33, pucx wrote:

enum.hasflag is not supported too.

lizhongchao13, 2019-04-10:
solved，modify code about IReadOnlyList，run successed， exported dxf file is also right。

closed 2019-04-10 by lizhongchao13