haplokuon/netDxf #58: Code error: font name does not cope from style

Jakobinec, 2018-08-31:
MTextFormatingOptions file line 269:

            if (this.fontName != null)
            {
                if (this.bold && this.italic)
                    formattedText = string.Format("\\f{0}|b1|i1;{1}", this.fontName, formattedText);
                else if (this.bold && !this.italic)
                    formattedText = string.Format("\\f{0}|b1|i0;{1}", this.fontName, formattedText);
                else if (!this.bold && this.italic)
                    formattedText = string.Format("\\f{0}|i1|b0;{1}", this.fontName, formattedText);
                else
                    formattedText = string.Format("\\F{0};{1}", this.fontName, formattedText);
            }
            else
            {
                if (this.bold && this.italic)
                    formattedText = string.Format("\\f{0}|b1|i1;{1}", this.style.FontFamilyName, formattedText);
                if (this.bold && !this.italic)
                    formattedText = string.Format("\\f{0}|b1|i0;{1}", this.style.FontFamilyName, formattedText);
                if (!this.bold && this.italic)
                    formattedText = string.Format("\\f{0}|i1|b0;{1}", this.style.FontFamilyName, formattedText);
            }

If I don't set the font in MTextFormatingOptions and it isn't italic or bold it will never copy from TextStyle

Jakobinec, 2018-08-31:
(last else in second block  is absent)

Jakobinec, 2018-08-31:
And I tryed to use .ttf file when I set TextStyle but in document such font wasn't.
Does library must embed font if I set the path when I created TextStyle?

haplokuon, 2018-09-10:
I will clean the MTextFormatingOptions. About the use of fonts in the TextStyle keep in mind that the program loading the file needs to know where to find them, under normal circumstances they should be installed in your system fonts directory.

closed 2018-10-17 by haplokuon