haplokuon/netDxf #66: Set image as background in the DXF file.

abhijitOffice, 2018-10-12:
I want to set an image as a background of the dxf file which I am creating. Other entities should be drawn on that image. Any suggestions on how to do it?

haplokuon, 2018-10-14:
To work with bitmaps you need to use the Image entity, just place it anywhere with the dimensions you want. Later, you can place any other entities like lines, polylines,... on the same plane. Keep in mind that the drawing entities are drawn in the order they were created. In the TestDxfDocument project there are lots of little samples on how to create and modify the available entities.

In a DXF there is no such thing as a background image, an image is treated like any other entity, properties like Color, Linetype, and Lineweight will control how the image border is drawn; so, there is no need to manually place a rectangle around the image limits.

Daniel

abhijitOffice, 2018-10-18:
Hello Daniel,
Thanks for your reply. I have changed the order of creating entities and it works. Thanks a lot.

Another thing is when adding the PNG images to the DXF file every image has a border that's making the PNG images non-transparent. And I want to show Images as it is. I mean transparent.

I have tried the Transparency option of the image but it's working like opacity.

Please give any suggestions on that.

haplokuon, 2018-10-19:
Are you talking about preserving the image alpha channel? If this is your problem, it is controlled by the DisplayOptions property of the Image entity, just add the flag TransparencyOn, by default it is off.

Daniel

abhijitOffice, 2018-10-22:
Hello Daniel,

Setting TransparencyOn makes the image transparent but the border around the image is still there in black color.

as shown in the image the image has a border drawn.
[image]

haplokuon, 2018-10-24:
Didn't you want to draw the border of the image? Nevermind, yes you can turn it off with
```c#
[your DxfDocument instance].RasterVariables.DisplayFrame = false;
```
This is a global variable, therefore it will affect all images in your drawing, you cannot turn them off individually.

closed 2018-11-23 by abhijitOffice