haplokuon/netDxf #70: Support for old R12/R13/R14 (question)

OakLD, 2018-11-23:
I wonder, how difficult it would be to include support for old R12/R13/R14 formats. As odd as it may seem, many new shiny 2018 CAE still generate DXF output in old R12 format...

I saved a simple file in ASCII R12 and 2010 and see it is of similar structure, though of course the 2010 file is much more complicated, having (for a simple file with a 1 circle) nearly 9 thousands lines, whereas R12 has only 1057 lines. I guess it also depends on SW which generates it and a template (I used DraftSight).
The data storing format is quite close (ENTITIES section and such), I guess it would be feature-removal task more than feature-adding, for the old formats.

Since I'm not C# fluent, I'd like to first ask if you (author) see it the same way and if you think it's feasible. I'd then look into the code in detail and try to elaborate on possible support.

haplokuon, 2018-11-27:
The difficulty will depend a lot on how familiar you are with the DXF format, if you are it is not all that hard it really is a matter of time.

Normally any modern program that handles DXF files should be able to read and write more up to date DXFs, try to look if there is somewhere in your software, in the general options, configuration or any other available options in the save dialog box, where you can change the DXF version.

Daniel

OakLD, 2018-11-29:
Thank you for your answer, I will look at it then. I wrote a DXF library myself many years ago, but it was quite primitive and featured only one format.
All I want to add is support for reading lines, arcs, circles and few of their properties (color, layer name). If I succeed I send you code changes, but I guess this scope is too limited in terms of features to include into officital release.

The thing is, that the DXF come from many different people and thus it is not possible to tell them all to set up their CAD/CAM/CAE output the (correct) way you suggested.