haplokuon/netDxf #71: Lineweight for Block

VadCher, 2018-11-29:
If EntityObject.Lineweight == Lineweight.ByBlock

Were can I find Lineweight in this case?

Vadim.

haplokuon, 2018-12-03:
It is missing, I will add it. On my defense I will say that it does not appear in the documentation. Something similar happens with the Color, Linetype, and a few other properties. I need investigate it further.

haplokuon, 2018-12-11:
I was wrong the code is ok. When a Color, a Lineweight, a Linetype, and/or Transparency is set ByBlock it means that it will inherit the properties of the block reference (Insert) not the actual Block definition. I used AutoCad for, I do not know how many years, and never used a ByBlock property.