haplokuon/netDxf #72: Inserts no longer contain items in paper space (only model)

EmiXikes, 2018-11-29:
I am not sure if this is by design, but here is my problem. 

In some earlier versions (2.0.3 for example), when I loaded a dxf file, the inserts collection would contain all the block references in the file, no matter if there were in Model space or Paper space (layouts).
In the latest version the Inserts collection now only contains items found in Model space.

If this is not a bug, but, in fact, by design, then how can I now access items (block refs, for example) found in Paper space?

haplokuon, 2018-12-03:
Yes I change it, in version 2.2 I reorganized how and where the entities data is stored. Now the memory footprint of the library should be somewhat lower. For more information check the sample method AccessModelBlock() in the TestDxfDocument project.

EmiXikes, 2018-12-11:
Thank you for the answer!
So now, if I'd wanted to iterate through all the inserts in an entire document, I'd have to do so on layout-by-layout basis?

haplokuon, 2018-12-12:
Yes. You can also access directly the entities of a Layout through its associated block and select the type you need, same thing. All those Lines, Inserts, .... properties of the DxfDocument do that through Linq. Hopefully this is not too much of a hassle, but I didn't want to have references to the same data in two different places, and under normal circumstances you will work in Layouts independently.