haplokuon/netDxf #73: Lineweight for Layer

ThomasStuevenScheugenpflug, 2018-12-03:
In **netDxf/netDxf/Tables/Layer.cs** Property **Lineweight** an exception is thrown if the Lineweight is set to "ByLayer" or "ByBlock". Unfortunately there is no comment in the commit that introduced this change why the lib is enforcing this rule. 
The way I read the DXF documentation both values would be perfectly legal (nonsense but legal). AutoCAD is able to load Files where the lineweight of a Layer is set to ByLayer.
I think it should be left to the Application to make sure it does not crash when finding this type of setting. The library should reflect the File as it is. 
 
Thank you for some clarifying words.
[image]

sorifiend, 2018-12-03:
Just my 2c worth:
Remember that you are trying to set the line weight property for a layer to reference itself, so as you say that is nonsense. This value should always be defined, and anything else would be undefined behavior or guess work (should it be 0.1 or 1 or maybe the average weight of all other lines in the drawing?), and NetDXF shouldn't be making that decision, the best case is that the user/coder of each individual program chooses something valid instead because they know what they need. 

As you know, the reason that it works in Autocad is because they "fix" it and default to something, and you can do the same and add a simple block of code that corrects a file on import to whatever line weight suits your needs.

You can always manually revert the file afterwards to have a bylayer line weight, but then the same undefined behavior will happen in the next program that opens the drawing, and you never know what it will show as.

haplokuon, 2018-12-03:
If the Load method finds a value ByLayer or ByBlock while reading the Layer information, it should fix it assigning the Lineweight.Default value to avoid the exception. I will fix that.

In any case no program should have written one of those values to the DXF, as @sorifiend has point it out it has no sense, and AutoCad does not allow them through its UI.

There are lots of values through the DXF that are actually restricted to some kind of range, and AutoCad what usually do is fixing them giving a default value. There are too many crappy DXF writers out there that write wrong values hopping that somebody else will fix them. And since the official DXF documentation its even worse, this will not be the last time one of this problems appears. Keep checking.

ThomasStuevenScheugenpflug, 2018-12-04:
Thank you for the quick response.
I know the value makes no sense and I have no Idea how the value got there in my file. It came from a customer. Changing it automatically to Lineweight.Default seems to be a good choice.

haplokuon, 2019-06-23:
Fixed in netDxf 2.3.0

closed 2019-06-23 by haplokuon