haplokuon/netDxf #74: Give Application the chance for detailed Errormessages

ThomasStuevenScheugenpflug, 2018-12-03:
Currently the Load-Method is swallowing all Exceptions but one. That prevents the Application from generating an Errormessage that actually helps the user fix the Problem. Right now the App can only tell Version-Problem and "something went wrong while loading the File".

Please give the App an opportunity (or is there one I did not see?) to generate detailed Messages. Either throw the Exceptions (bad for older Apps) or supply a (kind of) **GetLastError** Method.

haplokuon, 2018-12-03:
Sorry, but I will not implement that. Generating a proper error message is too much hassle for very little gain, too many variables for a file format so loosely structured as the DXF. Not even AutoCad does it, for old stuff is fine but once you go deep into the territory of the objects and handle references usually an error there leaves AutoCad with a hard crash or at least leaves it unstable.

If you want check the exceptions use the Debug build, under normal circumstances if loading a DXF fails there is nothing a common user can do. If you find a file that netDxf ia unable to load but other loader are like the TrueDwgViewer, I will need to check it most probably you found a bug.

ThomasStuevenScheugenpflug, 2018-12-04:
I understand the problem and agree.
Just to clarify: This request is somewhat related to issue #73. If I had the Exception in the Application I could have pointed the user to the crappy DXF File and even tell him what was wrong.
But I am aware that this would not be true for all errors in the files.
Thanks

haplokuon, 2018-12-04:
If you use the Debug build that exception will raise, other cases might not be so obvious.

closed 2018-12-04 by ThomasStuevenScheugenpflug