haplokuon/netDxf #75: Thickness is not applying for Line or LwPolyline

abhijitOffice, 2018-12-05:
Hi There,

I am trying to change the thickness of the line but it seems it's not working.

I have tried Thickness property of Line class also tried for LwPolyline class. When I see the dxf file in the editor the line shows same as other lines. (no thickness applied.)

A second thing is if I apply the Lineweight property the thickness is getting applied but we have Lineweight as Enum with a maximum value of W211(2.11 mm) and in my case, I have the thicknesses as in numbers as 6,8,10 etc.

So I just want to show the line thickness in any way. any suggestions?

Thanks in advance.

haplokuon, 2018-12-09:
The thickness do not represent the width of a line but its height along its normal, this is the nomenclature given by AutoCad very long time ago when its 3d capabilities were called 2 and a half.

The lightweight is what it is I cannot do anything about it, that is how it is defined in AutoCad. What you can do is always use LwPolylines, even if they are made of a single line, and use an arbitrary width for them.

haplokuon, 2019-10-23:
The LwPolyline width is set per vertex. The SetConstantWidth method is just there for convenience, it just iterates through the polyline vertices and set the the same with for all of them.