haplokuon/netDxf #76: netdxf  Can you mirror objects？

1603598052, 2018-12-13:


sorifiend, 2018-12-13:
Yes. Just choose an axis (Middle of your object?), then iterate all the points in your object and move the points accordingly with some simple math.

Netdxf does not come with a mirror/flip method, but the above solution is straightforward and should be very easy and quick to implement.

H-Ashrafi, 2022-05-26:
`        /// <summary>
        /// mirrors an entity 
        /// </summary>
        /// <param name="entity"></param>
        public static void MirrorX(this EntityObject entity)
        {
            //make the mirror matrix (determinant)
            var m3 = new Matrix3(-1, 0, 0, 0, 1, 0, 0, 0, 1);

            entity.TransformBy(m3, new Vector3());
        }

        /// <summary>
        /// get a list of lines, circles, ellipses, polylines, arcs, splines and light polylines
        /// </summary>
        /// <param name="document"></param>
        /// <returns></returns>
        public static IEnumerable<EntityObject> GetEntities(this DxfDocument document)
        {

            //make a list of EntityObject
            var ents = new List<EntityObject>();

            //get liens, circles and ...
            var lines = document.Lines.Cast<EntityObject>();
            var circles = document.Circles.Cast<EntityObject>();
            var elipses = document.Ellipses.Cast<EntityObject>();
            var polys = document.Polylines.Cast<EntityObject>();
            var arcs = document.Arcs.Cast<EntityObject>();
            var splines = document.Splines.Cast<EntityObject>();
            var lightPolys = document.LwPolylines.Cast<EntityObject>();

            //add them to the list of entities
            ents.AddRange(lines);
            ents.AddRange(circles);
            ents.AddRange(elipses);
            ents.AddRange(polys);
            ents.AddRange(arcs);
            ents.AddRange(splines);
            ents.AddRange(lightPolys);

            //return the result
            return ents;
        }

        /// <summary>
        /// mirrors lines, circles, ellipses, polylines, arcs, splines and light polylines
        /// </summary>
        /// <param name="document"></param>
        public static void MirrorEntities(this DxfDocument document)
        {
            var ents = document.GetEntities();
            foreach (var item in ents)
            {
                item.MirrorX();
            }
        }`