haplokuon/netDxf #80: Dimensions question

FabioVerzeni, 2019-01-17:
Hi

I have a question on the LINEAR DIMENSIONS

This is my situation:

I have a segment identified by the vertex V1 and V2.

When I add a LINER DIMENSION selecting the vertex V1 and V2 in this order. I have the data DimLinePosition (the location of the dimension line) with values that are the same of the position in the DXF file.

When I add a LINER DIMENSION selecting the vertex V2 and V1 in this order I have the data DimLinePosition with values that are not the same of the position in the DXF file. The vertex is rotated by 180 degrees and moved.

Now, to show correctly the dimension I have to rotate and move the DimLinePosition point. To do this is not a problem, my difficulty is to understand when to do the operation.

Can somebody give to my some details on this problem ?

Thank you

haplokuon, 2019-01-19:
When creating a linear dimension the offset value must be positive, and it is always measured to the positive side of the vector defined by the start and end points of the dimension reference line. Keep in mind that I am assuming a right handed coordinate system and rotations are positive counter-clock wise.

Therefore, given a positive value of the offset if you switch v1 and v2 the dimension line will be created in opposite sides of the reference line.

FabioVerzeni, 2019-01-21:
Thank you for the explanation

Can you tell me which data I have to use to understand if the DimLinePosition point is on the opposite side ?

See in the examples below the data that I have in the class netDxf.Entities.LinearDimension

First case. I select first the point 2;2 and after 2;6: 
FirstReferencePoint 2;6
SecondReferencePoint 2;2
DimLinePosition 3;2
Offset 1
Rotation 270

Second case. I select first the point 2;6 and after 2;2: 
FirstReferencePoint 2;6
SecondReferencePoint 2;2
DimLinePosition 1;6
Offset 1
Rotation 90

In the first case the DimLinePosition is in the correct side, in the second is on the opposite side.

I tried to use the angle between 0-180 is on the opposite side, 180-360 on th ecorrect side, but doesn't work in some case.

Best regards

haplokuon, 2019-01-26:
When the dimension rotation is 90 the dimension line will appear on the left, if it is 270 it will appear on the right. I was explaining it wrong, when I said in the last comment that the dimension line is drawn in the positive side of the line, this is only applicable to aligned dimensions, for linear dimension is the rotation value that defines where the dimension line is placed (0º up, 90º left, 180º down, 270º right) independently of the angle of the reference line.

If what you want is to draw a dimension parallel to the reference line use the AlignedDimension instead, you will not need to deal with the rotations.