haplokuon/netDxf #81: Text along an arc

steve997, 2019-01-24:
Is it possible to get NetDXF to write text along the arc of a circle?
Is there a property to retrieve the width of some text?

haplokuon, 2019-01-26:
No. There is no entity or parameter in the DXF text entities to adapt a text to an arc. You will have to do it manually. And your second questions is also no.

shawncurry, 2019-07-19:
There is an "AcDbArcAlignedText" entity.

```
100
AcDbArcAlignedText
  1
The Quick Brown Fox Jumps over the Lazy Dog
  2
Arial
  3
```

- In AutoCAD, draw an arc.
- Command:  ARCTEXT
- Select the arc.
- Enter some text.
-

haplokuon, 2019-08-06:
Let's put it this way there is no DOCUMENTED entity that describes a text along an arc. The DXF documentation lacks a lot of information. It can actually contains a lot more things like information saved by external plugins, that usually involves the CLASSES and OBJECTS sections, and the information available about them is too little, it is just a brief description of the codes not how they actually works.

AcDbArcAlignedText is not an entity but a subclass marker, entities are only the ones defined in the ENTITIES section.