haplokuon/netDxf #84: Project is currently not Strongly-Named

peSHIr, 2019-02-04:
I started out using the netDXF NuGet package, but as all our own assemblies are strongly named, our code would not compile as the NuGet package is not strongly named. Checking the sources on here on github, I find that the original project is not strongly named either. Could you please make the project strongly named, so as not to exclude projects that are strongly named from easily re-using this code without having to be forced to do own compilation and update/version managing in the future?

https://docs.microsoft.com/en-us/dotnet/framework/app-domains/strong-named-assemblies

haplokuon, 2019-02-26:
I never cared to much about it, and nobody asked about it before. I will look it to it. Nevertheless the source code will always be available in case of special needs.

peSHIr, 2019-02-26:
Yes, you're right. Was able to use the source code, in a namepace of my own
with a strong name of my own. Still, being about to reuse the proper
downloadable assembly with the "official" namespace would be preferable. If
you ever get round to it.

haplokuon, 2019-06-23:
The new binaries in the netDxf 2.3.0 release are now strongly named, hope this is what you need.

closed 2019-06-23 by haplokuon

peSHIr, 2019-06-23:
Will have a look on Monday. If this indeed enables me to simply use a NuGet
assembly reference instead of including the source code in an assembly of
my own I will switch in a heartbeat. Thanks in advance!

haplokuon, 2019-06-23:
Keep in mind that I do not publish the library through NuGet, all I do is here at Github. It is another user who maintains the NuGet package.