haplokuon/netDxf #87: Problems upgrading from 2.0.2 to 2.2.0.1

stefanbollmann, 2019-02-22:
I get errors when the offset of an aligned Dimension is negative. Negative offsets with Version 2.0.2 were o.k.
Font Type Arial brings error: Only True Type Fonts are allowed (although it is a truetype Font). Was also o.k with Version 2.0.2.

haplokuon, 2019-02-26:
Did you mean version 2.2.1 instead of 2.2.0.1?

A negative offset for an AlignedDimension is the same as swapping its start and end reference points. In this case it is trivial to allow for negative values, but I just wanted to keep all dimension offsets as non negative values, and simplify the internal calculations of the dimensions.

About your second question, I don't know, it should not be happening. I double checked that and the arial.ttf font that comes with windows should work just fine. Please, post some code on how are you initializing the TextStyle, so I can take a look at it, version 2.0.2 is very old.

stefanbollmann, 2019-02-27:
Thanks for your answer.

I use 2.0.2.0 at the moment.

I can change the points for the dimension, that's no problem.

The code for the Text Style:

DrawTextToBlock("Klemmenplan", Klemmenplan, xPos, yPos, 60, "Arial", netDxf.Entities.TextAlignment.BottomLeft);

void DrawTextToBlock(string myTextString, netDxf.Blocks.Block myBlock, double posx, double posy, double height, string  style, netDxf.Entities.TextAlignment textalignment)
        {
            netDxf.Entities.Text myText = new Text(myTextString, new Vector3(posx, posy, 0), height);
            myText.Style = new TextStyle(style);
            myText.Alignment = textalignment;                     
            myBlock.Entities.Add(myText);
        }

The Arial font is:
c:\windows\fonts\arial.ttf
Version 7.00

closed 2019-02-27 by stefanbollmann

haplokuon, 2019-02-28:
From your code:
`
myText.Style = new TextStyle(style)
;`
What is the value of the 'style' argument? Did you forget to add the extension .ttf by any chance?

stefanbollmann, 2019-03-01:
Yes, I used the Style without the extension. Thank you for your answer. I will try to update next weekend.