haplokuon/netDxf #88: Drawing Extents

MikeRensing, 2019-03-07:
Is there a was with netDxf to find the extents of the model? The min and max points for everything drawing in the model.

haplokuon, 2019-03-09:
No. There are some entities I cannot calculate, it like Text, MText, and Tolerance; I have no way of measuring strings based in TTF and SHX fonts. Also the Underlay and Shape entities have its size defined by the scale of its original dimensions that comes from the PDF, DWF, DGN, and SHX file.

For newly loaded files you can read from the DrawingVariables of the document.

OakLD, 2019-09-26:
I would just complete @haplokuon 's answer above, that in reality it is usually not an issue, it is easy to calculate bounding box with "engineering precision", meaning it will not get issues under any common circumstances.

HYMMA, 2021-02-10:
if this is just for drawings components like lines and circles you can use something like this
```
private double[] GetBoundingBox(DxfDocument document)
{
            var lines = document.Lines;
            var circles = document.Circles;
            var ellipses = document.Ellipses;
            var polys = document.Polylines;
            var arcs = document.Arcs;
            var splines = document.Splines;
            List<Vector2> vectors = new List<Vector2>();
            foreach (var line in lines)
            {
                var vector = new Vector2(line.StartPoint.X, line.EndPoint.Y);
                if (!vectors.Contains(vector))
                    vectors.Add(vector);
            }
            foreach (var circle in circles)
            {
                vectors.AddRange(circle.PolygonalVertexes(4));
            }
//continue the same logic for the rest
}
```

LeiYangGH, 2021-11-21:
for **line** and **circle** of course calculation is simple. but what about **arc** for example?

OakLD, 2021-11-21:
> 

For the arc, you have to check if it passes 0° for X max, 180° for X min, 90° for Y max and 270° for Y min. If that extreme point is passed the bounding point in that direction is center plus radius, otherwise it's max of Start and End point coordinate.