haplokuon/netDxf #89: Issue in dxf file Drawing

mohammedibrahimlj, 2019-03-08:
While i trying to draw file a line  using netDxf it shows the line diagram 90 degree reverse.

haplokuon, 2019-03-09:
Show something that I can actually reproduce.

mohammedibrahimlj, 2019-03-15:
the issue is exist while comparing the image from windows form picture and dxf file. Kindly assist me draw a bezier curve in dxf file please share the sample code to my mail address(Mohammedibrahim.lj@gmail.com) 
thanks

haplokuon, 2019-03-22:
The start and end coordinates of a line are what they are. How you draw them is another issue. My only guess with such a limited info is that you are drawing in one coordinate system while expecting another. For example, if you are using, GDI the origin is at the top-left corner with the X axis going from left to right and the Y axis from top to bottom.

About the bezier curves, sorry I cannot help you, and I do not answer privately, any code for netDxf will be published here. Check [this](https://github.com/gradientspace/geometry3Sharp) you might find something useful there.