haplokuon/netDxf #93: MLine limitations

SamuelLordBlack, 2019-03-12:
Hi. I just pulled down your latest release (06 March 2019), and am having trouble with MLine. The major issues I'm seeing are

1. An odd number of elements is not supported, so you can't draw things like a two-lane road with a centerline unless you define the centerline twice.
1. Explode() does not support filled MLines. If you have the actual boundary, perhaps generating a solid Hatch would work? (This is the approach I'm going to try.)
1. Arc endstyle on a multi-element mline style sometimes generate an internal circle (in the wrong direction and from an internal element rather than using the outer elements) rather than the correct end cap.

And thanks for the March update - it was nice to be able to rip out my implementation of Insert.Explode() in favour of yours.

SamuelLordBlack, 2019-03-12:
I found the fix for #3. There are two problems with arcs.

1. The indexing is incorrect when calculating the center. Here is one example (it needs to be fixed in four places - one for each arc end cap type). I have highlighted the fix.

`Vector2 center = Vector2.MidPoint(cornerVextexes[0][0], cornerVextexes[0][cornerVextexes.Length - 1]);`
should be
`Vector2 center = Vector2.MidPoint(cornerVextexes[0][0], cornerVextexes[0][cornerVextexes*[0]*.Length - 1]);`

2. The colors in AutoCAD don't work quite as you have them implemented. The start cap can be split color, but the end cap never is. In those two places, you should unconditionally use color2 and linetype2.

The attached diff addresses these situations, and the results match AutoCAD exactly.
[arcdiff.txt](https://github.com/haplokuon/netDxf/files/2958759/arcdiff.txt)

SamuelLordBlack, 2019-03-12:
As it so happens, this same fix takes care of issue number 1 in the original post as well. Just delete this code and it will work as the array indexing was the issue.

`if (this.style.Elements.Count % 2 != 0)
throw new InvalidOperationException("The number of distances for a given MLineStyleElement must be an even number.");`

SamuelLordBlack, 2019-03-13:
I found another bug. If the MLine style has start arc caps and the two lines have different colors, the position  of the junction depends on the start angle.

Change (in two places)
`double midAngle = startAngle + 90.0;`
to
`double midAngle = startAngle + 180 - this.style.StartAngle;`

SamuelLordBlack, 2019-03-14:
And to fix mirrored MLines, change (in four places)
`double startAngle = Vector2.Angle(start - center) * MathHelper.RadToDeg;`
to
`double startAngle = Vector2.Angle(this.scale * (start - center)) * MathHelper.RadToDeg;`

haplokuon, 2019-03-22:
Consider any code commit that does not correspond to a release as broken by default. I am still working on it. And posting issues like this is not helpful in any way, all your code proposals are already superseded. If you really want to help post a sample DXF or a sample code that I can test.

About the end round caps, I do not know why AutoCad does it that way, for me it looks like a bug. I will keep breaking the arcs in half if the color and/or the line type are different.

haplokuon, 2019-03-23:
About MLine with a fill color, when they are exploded in AutoCad that information is not used to generate any additional entity, but it is easy to create a solid Hatch with the boundary of the MLine.

haplokuon, 2019-06-23:
Fixed in netDxf 2.3.0

closed 2019-06-23 by haplokuon