haplokuon/netDxf #97: Radial/Diametric dimension prefix isn't displayed in AutoCAD

Ars3i, 2019-03-22:
Radial and Diametric dimensions don't display the "R" or "Ø" prefix. Tested in AutoCAD 2012 and 2016. 
It's probably caused by this sequence in AcDbDimStyleTableRecord:

**3
<>**

If I manually delete these two lines from a dxf file. The problem is gone.

Original:
[image]

Modified:
[image]

[dimension drawing.zip](https://github.com/haplokuon/netDxf/files/2996478/dimension.drawing.zip)

haplokuon, 2019-03-27:
Correct, no need to delete both lines just the <>. The DXF code 3 for the dimension style represents the combination of the prefix and suffix strings defined in the style. The characters <> are substituted by the actual dimension measurement but it is only required if the a prefix is present. I will fix it.

haplokuon, 2019-06-23:
Fixed in netDxf 2.3.0

closed 2019-06-23 by haplokuon