haplokuon/netDxf #103: block.Record.IsForInternalUseOnly issue

johnny0618, 2019-04-25:
For example there is a block named A, and under A entityCollection there is one Insert Block entity
A Block :[entity1]

When change block name and edit block in autodesk cad and save again ,it cause very randomly will to this

A Block :[]
*U107 [entity1]

Means the entity come from A Block Collection to some temp block start with* (IsForInternalUseOnly ). But in cad it is still under A Block Collection.

haplokuon, 2019-04-26:
I guess you are loading a DXF with netDxf that contains dynamic blocks, and they are not supported.

All blocks which name starts with '*' are used by AutoCad internally to represents different things, for example, the *D# are used by dimensions, the *T# by tables, and *U# are used by dynamic blocks. This is a copy of a discussion topic posted long ago in Codeplex that might clarify your doubts:

> #### Blocks are renamed randomly
>
> Hello... First thanks for the great effort
>
> I am using netDXF library to load a dxf file that contain some blocks... When I open the drawing in Autocad, I have a block called "D-700-b". This is a dynamic block with custom attributes.
> I use the following code to get the Block reference inserts as a List of DxfObject
>
> ```
> DxfDocument dxf = DxfDocument.Load(filePath);
> List<DxfObject> dxfObjects = dxf.Blocks.GetReferences("D-700-b");
> ```
> dxfObjects return an empty list... after debugging I figured out that the block has been randomly renamed to "*U56" ... So, I changed the code to the following:
> ```
> DxfDocument dxf = DxfDocument.Load(filePath);
> List<DxfObject> dxfObjects = dxf.Blocks.GetReferences("*U56");
> ```
> and dxfObjects return a list that contains 5 items which is true in the drawing
> 
> For the other blocks which are not dynamic blocks... there is no change in the original name
> 
> Do you have any idea why this is happening? Is there a workaround to get the original name? I want to be able to display the list of blocks with its' original name in a ListBox so that the user can pick his blocks by it's name
>
> Edit: I uploaded the DXF so that you can check it
> Thanks
> 
>    Check7_Walls_Arcs_doors_purged.dxf
>
> There is nothing wrong with it and it works as expected. What you are seeing is the way AutoCad handles dynamic blocks internally, netDxf will load them but you will loose all their dynamic parameters.
> 
> Imagine that you have a block with parameters, actions, and constrains. When you insert this block in the drawing it will point to it; but if, at any moment, you modified any of its parameters a new anonymous block definition will be created with the name *U#, where # is a positive integer. Additionally, its block record will store the handle of the original record in its extended data information.
>
> See the method DynamicBlocks() in the TestDxfDocument project, it will show you how to access the handle of the original block record from the extended data of those "*U#" blocks. If you have any more questions about this open a new topic in the discussion forum, since I will close this issue.
>
> Daniel
>
> 2016-06-06
>
> Thank you... very good explanation :)
>
> Mahmoud
>
> 2016-06-07
>
> Closed #13517 | Created 2016-06-04 | Updated 2016-06-10

johnny0618, 2019-04-27:
It works prefectly!!
By following your sample project,the block entity are found correctly.
Thank you so much for the help!

closed 2019-04-27 by johnny0618