haplokuon/netDxf #106: GDI+ Access violation caused by PrivateFontCollection in TextStyle.cs

pnjeffries, 2019-05-02:
When using netDXF to read a DXF file that includes TextStyle information in Unity, the application crashes due to an Access Violation in GdiPlus.dll GdipDeleteFontFamily().  This occurs in the built version of the Unity executable, but not when running in the editor.  This appears to be caused by the use of PrivateFontCollection within TestStyle.cs TrueTypeFontFamilyName. Commenting out the section of code (lines 375 to 382) using PrivateFontCollection prevents the crash.  I tried calling Dispose() on the PrivateFontCollection but this did not help.

haplokuon, 2019-05-02:
I guess your problem is related with [this](https://github.com/haplokuon/netDxf/pull/83). This is a user pull request merged to the master branch sometime ago, but still not included in the latest release.

pnjeffries, 2019-05-03:
Yes, it's the same bit of code, but the try-catch doesn't help in this case because the CLR [won't catch 'corrupted state exceptions' like access violations by default](https://msdn.microsoft.com/en-us/magazine/dd419661.aspx?f=255&MSPPError=-2147217396#Four).  One 'solution' would be to use the HandleProcessCorruptedStateExceptions attribute to force it to handle them, but this might still leave memory in a corrupted state, so probably not a good idea.  A better fix might be to [extract the font names from the TTF files directly](https://www.codeproject.com/Articles/2293/Retrieving-Font-Name-from-TTF-File) without using PrivateFontCollection.

haplokuon, 2019-05-04:
Well, if that was not your problem, I cannot help, I don`t have don`t use Unity.

haplokuon, 2019-05-21:
With the latest commit [2019/05/21] I removed the need to use the PrivateFontCollection class from the constructor of the TextStyle. See this [post](https://github.com/haplokuon/netDxf/issues/108).

13704017891, 2019-06-03:
@pnjeffries  Hello, I am also using this library in unity, but I don't know how to draw the model, can you refer to your project? I need some ideas.Thank you so much