haplokuon/netDxf #107: reduce size?

prgprgprg91, 2019-05-08:
Hi! What is allowable ways to reduce generated dxf file size? I already used LwPolyline instead Polyline. Is existed way to use single global style for ever entity?

haplokuon, 2019-05-09:
If you have not tried it save your DXF file as binary, it will always be smaller than its equivalent text version. Your other option is to remove any block, layer, text style, dimension style,... not used by any entity in your drawing, the equivalent to executing the purge command in AutoCad.

rdwh, 2019-11-14:
but how to remove not used entities by netDxf ?

haplokuon, 2019-11-18:
Use the RemoveEntity method of the DxfDocument class, this is the easy way.

Shakhmaev, 2020-01-29:
How to change text height for LinearDimension. this has 0.18 simplex value ALWAYS

haplokuon, 2020-02-01:
Answer me why question. Why did you make your question in an old post that has nothing to do with your problem?

Usually it is better to give as many details as possible about what you are trying to do, and obviously make your question in a new post, comment in existing ones if you actually have something to say about the discussion at hand.

My guess is that you are changing the TextHeight of the dimension style and later you add it to the document. Keep in mind that when you add entities to a document, its properties that refer to table objects like layers, line types,...; or in your case dimension styles will not overwrite the existing ones, this is intended.

When you create a dimension by default it is associated to the default dimension style called "Standard", but this one is also always present in the document, therefore, it will not overwrite it. You can create a new style for your dimension, directly modify the dimension style of the document, modify the text height after it has been added to the document, or you can use the dimension style overrides to individually change its properties.

If your dimension comes from an external DXF, set the dimension associated block to null to force AutoCad to rebuild it, you could use the existing methods inside netDxf to build dimension blocks but its functionality is limited.

Dig a little bit into the samples in the TestDxfDocument project.