haplokuon/netDxf #108: MTextFormattingOptions fails to report Bold, Italic or UnderLined properties when passes a TextStyle object

ThomasMahon, 2019-05-16:
It appears that the MTextFormattingOptions instance fails to report the properties of a TextStyle obtained from an MText entity. I have bold and italic MText entities and when I pass the TextStlye enitity extracted from the MText into the MTextFormattingOptions, the Bold and Italic properties report false.

I attach a sample file which demonstrates the problem.

[Text that is bold and italic.zip](https://github.com/haplokuon/netDxf/files/3188939/Text.that.is.bold.and.italic.zip)

haplokuon, 2019-05-21:
Yes, when both the font file name and the font information in the extended data (font family name and font style) of the TextStyle are present the font style was not being applied. After running a couple test, there is no information about the extended data in the TextStyle in the DXF official documentation, the font information always takes precedence over the font file value.

Therefore, I will also modify the TextStyle class to use the font file or the font family name plus the font style, but not both. to avoid having a font family with an style that do not correspond with the font file.

ThomasMahon, 2019-05-22:
Thanks for confirming - also thanks for the library, its awesome. For the timebeing I am using the raw XData to eastablish if the text is bold and/or italic but I look forward to your update. Also, it seems Underline property also doesn't work - I couldn't find any XData which refers to this property however.

haplokuon, 2019-05-23:
The code is already updated.

There is no option in the TextStyle for underline texts, however you can use the MText entity together with the MTextFormattingOptions, something along these lines:
```c#
TextStyle style = new TextStyle("Arial", FontStyle.Regular);
MText text = new MText() {Height = 10, Style = style};
MTextFormattingOptions format = new MTextFormattingOptions() {Underline = true};
text.Write("Sample text", format);
```
I suppose that AutoCAD internally requires to read the actual font file that corresponds to a given font family and font style to know its geometry, and if there is no file, it cannot draw a text with those options. Imagine that you have a font family "A", if that family does not have a file to represents, Bold for example, there will be no available font "Bold A". I am talking about the different font variations installed in your system not the font styles of the NET enum FontStyle.

The only way that I know to get the different variations of a font family and its corresponding font file is through the system registry, but at the moment I do not want to mess with it. So, the FontStyle enum shows the common options, but it will be your responsibility to know that a specific style is available for a given font family.

Autodesk is to lazy to provide a proper DXF documentation, there are too many things that I have to guess by pure trial and error, so problems may arise.

haplokuon, 2019-06-23:
Fixed in netDxf 2.3.0

closed 2019-06-23 by haplokuon

ThomasMahon, 2019-06-25:
Thanks Daniel, appreciated. I looked at the font registry technique myself, but it doesn't yeild the variants of a font, just the main font file. I saw some ways of doing it via StackOver flow but it was outside of the scope of my project so I didn't attempt to implement it - needless to say, I don't blame you for not going near it at this point in time.